udotdevelopment/ATSPM #9: Incorporate code to ignore stop bar sensors, use only Approach Count

udotdevelopment, 2017-08-31:
Arrivals on Red currently reads the stop bar and advance count sensors. This should look at advance counts only. Kittelson has written code to fix this and have posted it to OSADP.

udotdevelopment, 2017-08-31:
Approach Delay also currently reads the stop bar and advance count sensors. This should look at advance counts only. Kittelson has written code to fix this and have posted it to OSADP.

ian-cameron, 2017-09-11:
We did really change any code for this request, just edited the SQL database.  Pull request https://github.com/udotdevelopment/ATSPM/pull/15 changes the database setup, but if you already have setup your database you could run this SQL query:
    DELETE
    FROM [MOE].[dbo].[DetectionTypeMetricTypes]
    WHERE DetectionType_DetectionTypeID = 4 and MetricType_MetricID IN(8,9)

udotdevelopment, 2017-09-19:
Kittleson has submitted a fix for this through a pull request. This has been accepted into the master branch.

closed 2017-09-19 by udotdevelopment