udotdevelopment/ATSPM #17: Bulk Signal Import

Derek-Lehrke, 2017-11-29:
MnDOT recently got its site running and would like to know if there is a method for importing the basic signal information (ID, Primary Name, Secondary Name, IP, Lat, Log, Region, Controller) in order to start acquiring event logs without having to go through the GUI for every signal? Can I insert them directly into the "Signals" table?

udotdevelopment, 2017-11-29:
If you have an existing database you can import the signal information from there directly into the signals table. All of the approach and detector data has to be entered through the interface to ensure they are linked correctly unless you want to write an application to do it for you.  We wrote a conversion tool when we released 4.0 but it is specific to our database. If you have experience with dot net you could try to customize it to your needs.

aladavis, 2017-12-01:
GDOT pulled our signal list from our central system database to import into ATSPM. This took a little labor and data matching, but it was possible. For example, our central (Intelight MaxView) had all of the relevant information in three separate tables. Approaches and detectors still had to be entered as mentioned above. We used SQL Server Management Studio to execute the import via .csv

Derek-Lehrke, 2017-12-01:
MnDOT also has MaxView as its central system and I was able to assemble a SQL query to obtain the relevant data and export it to a CSV

What method did GDOT use for importing the CSV into the existing table?

aladavis, 2017-12-01:
We just imported a csv file into the signals table. We initially created a second table "Signals2" in the database and merged the new data into it. Once complete, we changed its name to "Signals" and deleted the old table. We then had to make sure the dependencies were reestablished, which was no issue.

Disclaimer: I'm a total SQL novice and this methodology is probably not a recommended approach by experts.

aladavis, 2018-04-18:
Would you be willing to share the query you all use to compare the MaxView database to ATSPM?

closed 2018-08-20 by udotdevelopment