udotdevelopment/ATSPM #21: FTPfromAllControllers Log

Derek-Lehrke, 2017-12-08:
Is there a built in tool for generating log files from the FTPfromAllControllers function?

ian-cameron, 2017-12-12:
There is a method we use to log the output of the program.  Its done by editing the scheduled task.  Under Properties of the scheduled task go to Actions, and click Edit.  Change Program/Script to:

    C:\Windows\System32\cmd.exe

and under Add arguments (optional): put in this, changing the path to your exe and desired output location:

    /c C:\SPMv4\FTP\FTPfromAllControllers.exe >> C:\SPM\FTP.log 2>&1

Hope that helps.

Derek-Lehrke, 2017-12-13:
Thank you for the reply. I did implement something similar but was more interested in whether there was additional debugging tools to increase the verbosity of the logs or adding the signal ID to the front of each line. Given that the FTP program is multi threaded I do not know what each line belongs to. 

Just in case anyone was interested the code I ran to save the log files was saved into a `.bat` file and the task schedule just calls it. With this method I get an individual log for each instance that it has been run.

`C:\AllFTPControllers\FTPfromAllControllers.exe >> "C:\FTPLogs\FTP_%date:~10,4%-%date:~4,2%-%date:~7,2%_%time:~0,2%_%time:~3,2%_%time:~6,2%.log" 2>&1`

closed 2018-08-20 by udotdevelopment