udotdevelopment/ATSPM #22: Server CPU/RAM requirements

Derek-Lehrke, 2017-12-08:
Does anyone have a list of CPU/RAM requirements per signal? 

I currently was allocated 4 CPU's (2.59GHz) and 8 GB of RAM and the system is struggling to retrieve and decode 400 signals in lest then 15mins. I could theoretically keep increasing these values since it is on a virtual machine but would like to know what others are running with to know what I should expect.

udotdevelopment, 2017-12-08:
We have 16 cpu's and 128GB of Ram for 2000 signals. We do not fully utilize our cpu but we use all of the RAM. Unfortunately it is largely dependent on how much data each intersection creates.

Derek-Lehrke, 2017-12-11:
What process is using so much RAM? MnIT is monitoring the virtual machine and the RAM usage is extremely low (2GB out of 8GB).  I am only collecting and decoding the logs right now.

udotdevelopment, 2017-12-11:
The decode and import process does use a lot of cpu depending on how much data is being imported. What most of our memory is used for is the sql server itself. Sql Server will claim as much memory as you will allow and that speeds the import process.

shanejohnson-udot, 2017-12-11:
At UDOT, the import process runs on a separate machine from the database server, as the import process can use up all of the available machine resources, and that would interfere with the database server.
The problem you are having, I think, stems from trying to decode and import too much data at once.  Try using the file-by-file decoder until the older files have been processed and deleted, then you can use the new decoder and import program.

closed 2018-08-20 by udotdevelopment