udotdevelopment/ATSPM #25: Controller Event Log Size

Derek-Lehrke, 2017-12-19:
My IT department has expressed concerns over the number rows in the Controller_Event_log table. Can anyone share there experience with backing up the table and/or performance issues querying the table.

Currently after roughly 3-4 weeks of data (~400 signals) it is at 50GB and 800 million rows.

udotdevelopment, 2017-12-19:
Unfortunately there is no way around this. The index takes care of the querying part pretty well but eventually you will reach a size that makes it difficult for your imports to keep up. We have created a partitioned table to handle this but you need an enterprise license to do it. Otherwise you will need to offload your data.

Derek-Lehrke, 2017-12-20:
Luckily MnDOT is using an Enterprise Licence of SQL2016 so the Partisan Feature is available to us. I am assuming UDOT used the 'Timestamp' column as the Key. Are you partitioning by the month?

udotdevelopment, 2017-12-20:
That is correct. We use the Timestamp column by month. I recommend creating your partition table under a temporary name and then renaming the controller_event_log table, rename your partition table to controller_event_log and then transfer the data from your old table. This allows minimal down time and allows you to continue to import your controller logs while you transfer the data from the old table. Here are the scripts we provided to GDOT to help them get their partition table started. ControllerEventLog.sql creates the table and AddPatition.sql shows how to add partitions.
[PartitionSql.zip](https://github.com/udotdevelopment/ATSPM/files/1576149/PartitionSql.zip)

closed 2018-08-20 by udotdevelopment