udotdevelopment/ATSPM #26: Link Pivot not working

lakecountypassage, 2018-01-15:
Has anyone had any problems with getting the Link Pivot to work on the website? I am getting the PCD diagrams for the intersections in the route list, and the information shows up in the SQL server under ApproachRouteDetail. On the website, it shows the processing icon for about 1 second, then doesn't display anything.

The ApplicationEvents table in the SQL server is not showing any Link Pivot errors. The only error I get is "Transfer Task Timed Out", but usually doesn't affect anything as the data looks complete when running all of the other performance metrics. It just may have missed a upload, but got the upload at the next scheduled time.

The web.config is referencing the correct file paths, DNS server and endpoint addresses. All other performance metrics for the traffic signals work without issue.

Are there typical modifications to the LinkPivot code that anyone makes?

aladavis, 2018-01-19:
We've also had issues with LinkPivot. Previous discussions with UDOT indicated that it was a bug somewhere - trying to recall what the specific issues were. I believe they were database issues. Our issue appears to occur somewhere within the \Views\ApproachRouteDetails\Create.shmtl

udotdevelopment, 2018-08-20:
This is fixed with the new route configuration in version 4.2

closed 2018-08-20 by udotdevelopment