udotdevelopment/ATSPM #27: ASC3 FTP - Don't retrieve latest file

Derek-Lehrke, 2018-01-22:
Has there been any discussion about implementing a function to skip the most current file on the ASC3 to avoid breaking up the 15 min ".dat" files and thus not having to turn the high resolution logger on and off. I have noticed at several MnDOT signals we are missing 15 mins of data which I am assuming is caused by the ON/OFF procedure in that it received the OFF command but not an ON.
 
I see that on line 336 the code has a list of the remote files with `.dat`

https://github.com/udotdevelopment/ATSPM/blob/cbdea9de6c6d1da8bc836d3218622ce847dc6c8b/MOE.Common/Business/Signal.cs#L336

So I was thinking something along the lines of:

```
if  (RemoteFiles > 1)
  OrderbyName  //Not exactly sure how to code this yet
  SkipTopFile  //or this

```
Based on line 397 the code only deletes files in the list made on line 367 so the the current file should be left alone.

Thoughts?

shanejohnson-udot, 2018-01-22:
Yeah,we thought about it.  There are some good reasons to do it that way.  
Ultimately, our engineers want the data as fast as they can get it.  the 10-15 minute delay they have now makes them grumpy.  So, in an effort to get them the most current information possible, we pull the active file.

udotdevelopment, 2018-08-20:
This pull request has been implemented.

closed 2018-08-20 by udotdevelopment