udotdevelopment/ATSPM #33: Siemens controller doesn't always let FTP user delete the logs from controller after import

ian-cameron, 2018-04-09:
Its a firmware problem that made the log files be owned by root and unable to be deleted by the admin account used to log in to FTP, and the FTP/Decoder would just re-import them, and end up with duplicate records in the ControllerEventLogs table.  The site seems to still work with the duplicate events, but the database grows much faster, and probably is slower because of the extra logs.  We only pull logs once per day so with 5 days of logs kept by default on the controller we had about 5x overhead in data, but if you pulled logs every 15 minutes and had this problem you'll end up with something like you data being duplicated up to 480 times.

ian-cameron, 2018-04-09:
We worked around this firmware issue by 1) skipping files that had all their records already in the database, and 2) importing records from current files that only exist after the last observation.  This fix may also be useful in other situations because it will not assume that all downloaded logs are current.  

This also allows you to not have to delete records from the controllers, which is useful if your engineers are pull the logs outside of the ATSPM software, such as for other analysis.

udotdevelopment, 2018-08-20:
The pull request for this has been incorporated.

closed 2018-08-20 by udotdevelopment