udotdevelopment/ATSPM #35: Purdue Phase Termination chart many unknown events

ian-cameron, 2018-04-09:
Siemens controller logs events at the same timestamp, and the way they're sorted by the software (in descending order) causes a lot of Unknowns (where event code 7 is immediately followed by another event code 7):
[image]

If you sort Ascending, then the unknowns don't happen:
[image]

Let me know if you would like further clarification

udotdevelopment, 2018-08-20:
This is fixed in 4.2.

closed 2018-08-20 by udotdevelopment

ian-cameron, 2019-08-09:
It appears this got 'unfixed' [line 91](https://github.com/udotdevelopment/ATSPM/blob/067083b13e62a910f1ed245b7260078d1c78e997/MOE.Common/Business/AnalysisPhase.cs#L91) with [this merge](https://github.com/udotdevelopment/ATSPM/commit/0d7b2412064c57a4e796410fdbc10f34046d874d)  that made it into the latest release and the current master branch.  We updated to release 4.2 and saw many unknowns again.   Removing the sort by descending, and leaving it ascending so that events 4 5 and 6 come before 7 eliminated the many PPT unknowns.   [Example](https://github.com/kittelson/ATSPM/commit/5572e91f3a8d54854fb5f772c8e4bec6e61fbd03)