udotdevelopment/ATSPM #38: Econolite Cobalt EOS

Derek-Lehrke, 2018-05-10:
Not sure if anyone has used the ATSPM software on a Cobalt running EOS. It would appear that Econolite now renames complete files as ".datZ" and the current one is a ".dat" 

The FTP process works fine and retrives them but the decoder does not. Not sure if you were going to make a separate decoder for Cobalts or not or wanted to keep it as part of the ASC3 Decoder.

https://github.com/udotdevelopment/ATSPM/blob/cbdea9de6c6d1da8bc836d3218622ce847dc6c8b/NEWDecodeandImportASC3Logs/Program.cs#L79

shanejohnson-udot, 2018-05-10:
Very interesting, thank you.
UDOT doesn't have EOS controllers yet.  However, the decoder works with Cobalt controllers on the ASC3 and 2070 platforms.
As far as I know, UDOT does not _currently_ have plans to add an EOS-specific decoder.

wgortney08, 2018-05-10:
> EXTERNAL EMAIL: This email originated from the Internet. Do not click any images, links or open any attachments unless you recognize and trust the sender and know the content is safe.
>
> MnDOT did deploy its first ATC intersection with a Cobalt running EOS so I will be trying to get it working. My thought is to just copy the "NewDecodedandImportASCLogs" and run it separately.
>
> I tried a couple of quick changes "**.dat*" and a new encoder with "*.datZ" and it did do any thing. I haven't dug into it too much but will over the next couple of weeks.
>
> [image]<https://user-images.githubusercontent.com/34098526/39889469-96890152-545d-11e8-8071-0d996e926c71.png>
>
> I added an example file if anyone feel like looking at it. I added the ".log" so github would load it
>
> ECON_10.69.157.16_2018_05_10_1200.dat.log<https://github.com/udotdevelopment/ATSPM/files/1992928/ECON_10.69.157.16_2018_05_10_1200.dat.log>

If you need the EOS decoder, you can contact me at Terry.Gortney@charlottenc.gov<mailto:Terry.Gortney@charlottenc.gov>

I have been the test guinea pig for EOS for a few years now.

Thank you,

Terry Gortney, P.E.
Senior Engineer - Signal Systems
Charlotte Department of Transportation

Chris-Sobie, 2018-05-11:
We are running into this issue in Albuquerque. We have the new Cobalt EOS decoder, and it's able to decode both .dat and .datz files. My thought was to swap it out with the existing "NEWDecodeandImportASC3Logs", but that didn't work. The decoder we received came with 1 .exe file and 2 .dll files. I'm relatively new to development so if anyone can lay out the steps to get the EOS firmware working, I would greatly appreciate it.

Thanks!

Chris Sobie
csobie@lee-eng.com

shanejohnson-udot, 2018-05-11:
"NEWDecodeandImportASC3Logs" is NEW because we wrote the translator directly in the program.  That avoids the need to turn .dat files into .csv files and read the .csv files into the database.  It reads the .dat files and converts them in memory, then imports the records to the database directly.  UDOT has 1000+ asc3 controllers, so it was well worth our time to make that effort.

If we have to use external tools, like the EOS decoder, then the steps are similar to all of the "decode and import" programs we have created for the other controller types (and you should probably look at those for examples).

Roughly, here are the steps:
for-each through the directories that hold the log files.
for-each through the log files and use the EOS decoder to create a .csv file
Go back through all of the .csv files, add the signalID to each row (usually we get that from the directory name)  This will create a complete event record.
Bulk-import the records into the database.
Delete the files (both the .dat and the .csv)

Chris-Sobie, 2018-05-24:
Hi,
We are working on the .exe.config file for setting up the EOS decoder for ATSPM.  We are using the following code to test the .config file but are having trouble getting the files to translate successfully.  Here is the code we are currently using with "LogPath", "DecoderPath","LogsPath" and "CSVOutPath" pointing to local folder on the server to test the process.  It would be great if someone can double check the code for any obvious omissions:

<?xml version="1.0" encoding="utf-8"?>
<configuration>
  <configSections>
    <sectionGroup name="applicationSettings" type="System.Configuration.ApplicationSettingsGroup, System, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089">
      <section name="DecodeSiemensLogs.Properties.Settings" type="System.Configuration.ClientSettingsSection, System, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089" requirePermission="false" />
    </sectionGroup>
    
    <section name="entityFramework" type="System.Data.Entity.Internal.ConfigFile.EntityFrameworkSection, EntityFramework, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089" requirePermission="false" />
  </configSections>
  <connectionStrings>
    <add name="DecodeSiemensLogs.Properties.Settings.SPM" connectionString="Data Source=localhost;Initial Catalog=MOE1;User ID=atspm;Password=LeeEng1" providerName="System.Data.SqlClient" />
  </connectionStrings>
  <startup>
    <supportedRuntime version="v4.0" sku=".NETFramework,Version=v4.5.2" />
  </startup>
  <applicationSettings>
    <DecodeSiemensLogs.Properties.Settings>
      <setting name="LogPath" serializeAs="String">
        <value>C:\Users\Administrator\Desktop\TestTranslator\</value>
      </setting>
      <setting name="WriteToConsole" serializeAs="String">
        <value>True</value>
      </setting>
      <setting name="DecoderPath" serializeAs="String">
        <value>C:\EOS\EOS\HighResTranslator\</value>
      </setting>
      <setting name="DeleteFiles" serializeAs="String">
        <value>False</value>
      </setting>
      <setting name="TimeOut" serializeAs="String">
        <value>1000</value>
      </setting>
      <setting name="LogsPath" serializeAs="String">
        <value>C:\Users\Administrator\Desktop\TestTranslator\</value>
      </setting>
      <setting name="CSVOutPAth" serializeAs="String">
        <value>C:\DecodedSiemensLogs\</value>
      </setting>
      <setting name="DeleteFile" serializeAs="String">
        <value>True</value>
      </setting>
      <setting name="forceNonParallel" serializeAs="String">
        <value>False</value>
      </setting>
      <setting name="MaxThreads" serializeAs="String">
        <value>10</value>
      </setting>
      <setting name="DestinationTableName" serializeAs="String">
        <value>Controller_Event_Log</value>
      </setting>
      <setting name="EarliestAcceptableDate" serializeAs="String">
        <value>2010-01-01</value>
      </setting>
      <setting name="BulkCopyBatchSize" serializeAs="String">
        <value>5000</value>
      </setting>
      <setting name="BulkCopyTimeOut" serializeAs="String">
        <value>10</value>
      </setting>
    </DecodeSiemensLogs.Properties.Settings>
  </applicationSettings>
  <entityFramework>
    <defaultConnectionFactory type="System.Data.Entity.Infrastructure.SqlConnectionFactory, EntityFramework" />
    <providers>
      <provider invariantName="System.Data.SqlClient" type="System.Data.Entity.SqlServer.SqlProviderServices, EntityFramework.SqlServer" />
    </providers>
  </entityFramework>
  <runtime>
    <assemblyBinding xmlns="urn:schemas-microsoft-com:asm.v1">
      <dependentAssembly>
        <assemblyIdentity name="System.Web.Mvc" publicKeyToken="31bf3856ad364e35" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-5.2.3.0" newVersion="5.2.3.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="System.Web.WebPages" publicKeyToken="31bf3856ad364e35" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-3.0.0.0" newVersion="3.0.0.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="Microsoft.Owin" publicKeyToken="31bf3856ad364e35" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-3.0.1.0" newVersion="3.0.1.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="Microsoft.Owin.Security" publicKeyToken="31bf3856ad364e35" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-3.0.1.0" newVersion="3.0.1.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="Microsoft.Owin.Security.Cookies" publicKeyToken="31bf3856ad364e35" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-3.0.1.0" newVersion="3.0.1.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="Microsoft.Owin.Security.OAuth" publicKeyToken="31bf3856ad364e35" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-3.0.1.0" newVersion="3.0.1.0" />
      </dependentAssembly>
      <dependentAssembly>
        <assemblyIdentity name="Newtonsoft.Json" publicKeyToken="30ad4fe6b2a6aeed" culture="neutral" />
        <bindingRedirect oldVersion="0.0.0.0-6.0.0.0" newVersion="6.0.0.0" />
      </dependentAssembly>
    </assemblyBinding>
  </runtime>
</configuration>

shanejohnson-atspm, 2018-06-21:
Chris,

Changing the .config file for an existing decoder program probably is not going to work.  Most likely, you are going to have to write a new program.  using one of the existing decoder programs as a pattern will probably help.

Chris-Sobie, 2018-06-21:
Shane,

I appreciate the email! Thank you.

Christopher Sobie, EI
SR Engineering Designer
Lee Engineering, LLC
(505) 338-0988 phone
(414) 412-7854  mobile
csobie@lee-eng.com<mailto:csobie@lee-eng.com>

gmonk, 2018-08-17:
The EOS datZ format is nothing more than a zlib comressed dat file.   So if you take the dat file and run it through binwalk or zlib_flate on a linux workstation you will end up with a dat file that can be decoded

binwalk -eM ECON_172.22.251.14_2018_08_16_1700.datZ  you will have a folder with _ECON_172.22.251.14_2018_08_16_1700.datZ  inside two file 0.zlib and 0    the 0 file is the dat file
or  using zlib_flate

zlib-flate -uncompress < ECON_172.22.251.14_2018_08_16_1700.datZ  > exctracted.dat  this will be the dat file you need

udotdevelopment, 2018-08-17:
We are meeting with Traffop on Monday to review their changes to the asc3 decoder. They have adjusted it to allow for the new format.

mandrewbot3k, 2018-08-17:
Just an FYI, our new Rackmount Cobalts are shipped with SW version 32.66.10, which also creates datZ files. I believe 32.65 does not, we are going to jump back in version to verify. My last 4 rackmount Cobalts I have received are running 32.66.10 though. 

The interesting behavior though is it does save .dat files still, but it's intermittent behavior. I'm only bench testing so it's hard to tell what is really going on. I suspect that it only creates the dat when there is something actually logged. See below.

[image]

I'm very new to ATSPM and am utilizing the raspberry pi data logging method to do a pilot program here, so maybe I'm missing something.

Traffop, 2018-08-20:
We have added support for decompressing the datZ files into the original decoder file in the last pull request. We use slightly different code in production therefore we recommend testing these changes before pushing it into your production environment.

dbrandesky, 2019-07-18:
I just tested the "NEWDecodeAndImportASC3Logs.exe" package that comes with SPM 4.2.0, and can confirm that it works with .datZ files from an ASC/3 running 2.66.30 software. I'll give Eos a shot sometime soon, but I would expect success there as well.

mandrewbot3k, 2019-07-18:
I just wanted to add in regards to the hi-res MOE logging on the ASC/3 platform, to make sure you are not running 32.66.10 with one minute hi-res MOE enabled (MM-1-6-1). 

Upgrade to 32.66.20 or using a different interval should work. When set to one-minute, the controller may freeze up. I found this in a benchtop setup when my ethernet was disconnected, It appeared to not to buffer correctly when the logs were not being downloaded, despite circular being enabled. However, the release notes for 32.66.20 state this issue to be corrected, though I have not verified yet personally. 

Earlier versions (such as 32.65.40) with logging enabled via SNMP or 32.66.20+ through the Logging menu (MM-1-6-1) should work fine otherwise. 

We are running Rackmount Cobalts for reference. Cheers.

dbrandesky, 2019-07-19:
The NEWDecodeAndImportASC3Logs.exe did not work with Eos log files, though the FTP uploader executable faithfully retrieves the files. I'm recalling now that there are some differences in the MOE data format for Eos, so that may take some dev work. If I have the opportunity I'll see if I can figure something out, but I am far from a Windows ASP.NET or JS developer so I may not be much help!

shawngotfredson, 2020-03-09:
Which version of the FTP program should I be using with the newer Cobalts? The one I am using works fine with older ASC3 firmware but it is not able to delete files off the controller it downloads the .datz files.

dbrandesky, 2020-03-10:
@shawngotfredson I don't know of any issues deleting files on Cobalts, and I don't know of different versions of the FTP application for Windows. If you haven't already, you might try connecting to the controller using the actual command-line FTP client and see if you can manually delete .datZ files.

shawngotfredson, 2020-03-10:
I am able to delete the .datz files with a Windows FTP client. It is when I am trying to run FTPfromAllControllers that I get an issue. I am running ASC3 32.67.20. 

It appears the directory structure may have changed as well on the newer firmware. I have to use /econolite/set1 vs just /set1. If I use just /set1 with the new firmware I get an error saying directory not found.

There is a chance I am using the wrong version of FTPfromAllControllers or I need to wait till development catches up. OR I maybe missing something simple. I have the software running on my older firmware so no worries.