udotdevelopment/ATSPM #40: IIS/MS SQL alternatives...

mandrewbot3k, 2018-08-01:
I am just curious if there has been any discussion among the user community about porting development for use on a Linux nginx server with the dotnet core or using alternate database platforms such as PostgreSQL? 

Obviously licensing is always an issue, so I'm just curious.

shanejohnson-atspm, 2018-08-03:
A few agencies have asked me about that.
Most of the data access for the ATSPM system is accomplished through entity framework.  there are a few tutorials that cover how to convert a project from using an MSSQL server to a different data source.  This one in particular might interest you: https://www.codeproject.com/Articles/783552/Using-PostgreSQL-with-Entity-Framework-in-ASP-NET

I won't pretend that it is going to be that easy, but it probably isn't as daunting a task as it first seems.

mandrewbot3k, 2018-08-04:
Thanks Shane. I’ll check it out!

closed 2018-08-20 by udotdevelopment

dbrandesky, 2019-07-19:
@mandrewbot3k were you able to make any headway with this? I'm very curious to see if this is possible so I can save a Windows license key...

mandrewbot3k, 2019-07-19:
@dbrandesky I was unable to find the time for the most part to get it to work properly. From my bit of research, the biggest issue was I was not confident that I wouldl be able to get the .NETcore packages to work as there is a specific requirement for version 4.5.2, and the .NET core packages for linux are limited to version 2 (with a version 3 preview out now). 

I think a full port would be required to use Mono in lieu of .NET core, since WCF and the web stuff is not integral into it. MONO does have a guide for this, but as a Civil engineer and not a software engineer, I did not have the time to take on this task, as my programming skills are limited, especially in the C# world. https://www.mono-project.com/docs/web/porting-aspnet-applications/ From my brief look, it seems as though you would also have to use PostgreSQL as well. 

Good luck, I'm definitely interested if someone does take this on.