udotdevelopment/ATSPM #42: Issue installing ATSPM 4.2

MorenoJu, 2018-11-27:
Hi all,
When installing ATSPM 4.2, I ran into the following error:

Unable to update database to match the current model because there are pending changes and automatic migration is disabled. Either write the pending model changes to a code-based migration or enable automatic migration. Set DbMigrationsConfiguration.AutomaticMigrationsEnabled to true to enable automatic migration.  

Do you have any suggestion on what to do?

Thanks

udotdevelopment, 2018-11-27:
Let me take look. Can you tell what the last entry in the _MigrationHistory table is?

udotdevelopment, 2018-11-27:
Also would you mind telling me what the name of the primary key on your signals table is?

udotdevelopment, 2018-11-28:
Ok. I think I have it fixed now and I have updated the executable for the database conversion on GitHub. Can you try it again?

MorenoJu, 2018-11-28:
Sounds good. We’ll get back to you when we’ve tried it. Thanks!

The information contained in this e-mail message is privileged and confidential and is for the exclusive use of the addressee. The person who receives this message and who is not the addressee, one of his employees or an agent entitled to hand it over to the addressee, is informed that he may not use, disclose or reproduce the contents thereof, and is kindly asked to notify the sender and delete the e-mail immediately.

MorenoJu, 2018-11-28:
Hi,

We tried the new code and yes, it is fixed. It created the database with content in the tables. We can continue with the rest of the installation now.

Thanks for the quick response!

MorenoJu, 2018-12-07:
We downloaded the updated version and we were able to continue with the installation.
Thanks for the support.

closed 2018-12-07 by MorenoJu