udotdevelopment/ATSPM #43: Sample data for ATSPM

MorenoJu, 2018-12-07:
Hi all,
I've recently installed ATSPM in my laptop (with some support from this forum, thanks! :) )
I would like now to have some actual data in the DB that the installation process created. Maybe some simulated data so I can try the import tool?
I was wondering if that's something someone in this community could send me for test purposes.
Thanks much!
Juan

taylormd3, 2018-12-07:
You can go to the UDOT ATSPM website at: http://udottraffic.utah.gov/atspm. Create an account.  Email marktaylor@utah.gov and inform me that you have created an account so I can assign you data as a user. You can then go to "Admin" and "Raw Data Export" and download the raw data files for various signals and save them in your database. You can also go to "Reports" and "Configuration" to see how to configure the data. I hope that will work. Let me know if you need help. Good luck!

closed 2019-08-05 by yunongxing

romik280, 2021-06-01:
> Вы можете перейти на сайт UDOT ATSPM по адресу: http://udottraffic.utah.gov/atspm . Завести аккаунт. Напишите по [адресу marktaylor@utah.gov](mailto:marktaylor@utah.gov) и сообщите мне, что вы создали учетную запись, чтобы я мог назначить вам данные как пользователя. Затем вы можете перейти в «Администрирование» и «Экспорт необработанных данных», загрузить файлы необработанных данных для различных сигналов и сохранить их в своей базе данных. Вы также можете перейти в «Отчеты» и «Конфигурация», чтобы увидеть, как настроить данные. Я надеюсь, что это сработает. Дай мне знать, если тебе понадобится помощь. Удачи!

Hello. I need data by incidents of roads. Is it in open access?