udotdevelopment/ATSPM #44: ATSPM 4.2 Webconfig File

Chris-Sobie, 2018-12-13:
I was working on a new install of 4.2 by following the ATSPM 4.2 Installation Manual documentation. When editing the web.config file, the documentation calls for updating the ImageLocation, LinkPivotImageLocation, and SPMImageLocation keys. However, these are not included in the code anymore. Is this intentional or should it match the code from 4.0.1?

shanejohnson-atspm, 2018-12-13:
Those values are in the database now and can be configured using the web ui.

udotdevelopment, 2018-12-13:
It is now in the general settings of the website. If you are following the installation manual it is step 2.1.6 b.

amoshouqx, 2019-03-06:
> It is now in the general settings of the website. If you are following the installation manual it is step 2.1.6 b.

Excuse me, I have the same question and this answer makes me even more confused. So does it mean that the website could be launched before setting the keys? Since there were no more settings for keys in the web config file, I skipped step 2.1.6 b,c,d and launched the website. However it failed(see the figure below). So could you please explain what I should do to solve this problem in details since I did get any idea from this answer. Thanks very much!

[image]

udotdevelopment, 2019-03-06:
The website can be launched without the image location set. That value is only used when a chart is run.

amoshouqx, 2019-03-06:
> The website can be launched without the image location set. That value is only used when a chart is run.

Thank you! So what could be the potential reasons for the error "A network-related or instance-specific error occurred while establishing a connection to SQL Server" when trying to launch the website?

udotdevelopment, 2019-03-06:
There are lots of potential configuration issues it could be but the first thing I would check is to make sure your database user is set up properly with the correct permissions. After that make sure your database is visible from your web server. This especially important if your database is on a separate machine and potentially behind a firewall.

amoshouqx, 2019-03-06:
Hi, I have checked the connectivity to the database using the login information set in the SQL Server Management Studio. It showed that the connection could be built up successfully. Since I set up everything on the same machine, I think it indicates the database is visible from my web server. I also followed everything step by step in the installation manual. So is there any other possible reason for failing to launch the website?

[image]

udotdevelopment, 2019-03-06:
Does your database user have admin privileges? When running for the first time it creates a lot of things that require admin. Also what version are you installing and is it a new install or an upgrade?

amoshouqx, 2019-03-06:
> Does your database user have admin privileges? When running for the first time it creates a lot of things that require admin. Also what version are you installing and is it a new install or an upgrade?

Excuse me, I do not know if I understand the question database "Does your database user have admin privileges" correctly. My answer would be that I am logging on the computer with an administrative account.  I am installing ATSPM version 4.2 which was downloaded from OSADP and it is a new install

Chris-Sobie, 2019-03-11:
I believe there were some files that were fixed after the release on OSADP, but I'm not sure they were incorporated into the OSADP package. Do we know if FHWA updated the release?

udotdevelopment, 2019-03-11:
For now I am advising people to get the release from GitHub. I emailed OSADP a few times and did not get a response but I have direct control of the GitHub release mechanism.

Chris-Sobie, 2019-03-11:
I haven't tried to install from Github. Do I just download the ATSPM.Executables.zip file and not the Source Code?

udotdevelopment, 2019-03-11:
https://github.com/udotdevelopment/ATSPM/releases

closed 2019-08-05 by yunongxing