udotdevelopment/ATSPM #45: No endpoint listening at MetricGenerator.svc

amoshouqx, 2018-12-17:
I am still installing ATSPM v4.0.1 and I am now going on the step 2.7 "Create Web Services".  I have finished the step from 2.7.a to 2.7.d. However, when I was trying on step 2.7.e by generating the charts on my localhost atspm website. The charts did not come out and there was an error message saying "**There was no endpoint listening at http://mysite.com/MOEWcfService/MOEWcfServiceLibrary.MetricGenerator.svc that could accept the message. This is often caused by an incorrect address or SOAP action. See InnerException, if present, for more details**". I checked all the steps above and I was pretty sure I did not made major mistakes. So why does this problem happen and what should I do to solve it? I am looking forward to your reply and thanks!

udotdevelopment, 2018-12-17:
It appears that it cannot resolve the address you have provided for the web services or the web services are incorrectly configured so it cannot generate the report. I would recommend browsing to your web services on the IIS Manager and seeing if they display correctly. If they do they will also display the address they are expecting to be contacted at which you can copy and paste into the web config file for your website.

amoshouqx, 2018-12-17:
I have checked my IIS Manager and it seems that the services are correctly displayed. I just wonder what can I do next?

udotdevelopment, 2018-12-17:
Did you copy the address of the service to your web.config? For example if your wcf service library displayed this http://MySite/MOEWCFLibrary/MOEWcfServiceLibrary.MetricGenerator.svc?singleWsdl you would copy http://MySite/MOEWCFLibrary/MOEWcfServiceLibrary.MetricGenerator.svc and put it in your web.config.

amoshouqx, 2018-12-17:
Yes. The installation manual has already mentioned that in step 2.7.4.c. I checked the web.config file again and verified that I did.

amoshouqx, 2018-12-17:
Is there any other likely reason for my problem? Thanks!

amoshouqx, 2018-12-17:
Excuse me for that  I do not know much about WCF Web Service. However, is the file "MOEWcfServiceLibrary.MetricGenerator" supposed to be like this?:
<%@ ServiceHost Service="MOEWcfServiceLibrary.MetricGenerator" %>

So the file only contains one code to serve the request of generating metric charts in my localhost website?

I checkd all I did for step 2.7 and I did not find anything wrong though...

udotdevelopment, 2018-12-20:
Sorry I did not follow your previous question.

closed 2019-01-18 by amoshouqx

Crear12, 2021-05-05:
> Is there any other likely reason for my problem? Thanks!

I met this error too. I had one installed successfully, but the new one failed. I'm not sure if it's because I didn't downgrade .NET back to 4.5.2 on this new server.