udotdevelopment/ATSPM #46: "EntityValidationErrors"

Chris-Sobie, 2018-12-20:
When trying to create a chart I receive the message "Validation failed for one or more entities. See 'EntityValidationErrors' property for more details." Restarting the server resulted in the ability to create a few charts, but the error returns. Also, I've had this same message pop up when I already generated a metric and scrolled the page. I've done some online research and can not seem to find out a solution to this. 

I am operating ATSPM version 4.0.1. 

Thanks. 

[image]

udotdevelopment, 2018-12-20:
4.0.1 can be a bit hard to troubleshoot because I don't have version I can try to replicate the problem with. In general entity validation errors refer to some missing piece of data that the program is expecting to be able to run the chart. I am not sure why this would change after restarting the server. You may want to check the set up on your signals just to verify everything required is present. You can also check the ApplicationEvents table in the database to see if any additional information about the error was recorded in the table.

udotdevelopment, 2018-12-20:
Does it happen with all chart types?

Chris-Sobie, 2018-12-20:
Yes, it happens with all chart types.

Chris-Sobie, 2019-02-15:
On a different ATSPM platform that is functional for running metrics, I am trying to update my detector setup for a signal. When I attempt to save the updates in the signal setup I am getting the same error: "An error occured while updating the entries, See the inner exception for details". I'm not sure why this is happening. Any ideas?

shanejohnson-atspm, 2019-02-15:
Try making one change and then saving it before moving to the next one.  I think the problem is you are inadvertently creating multiple detectors with the same detector ID.

Chris-Sobie, 2019-02-15:
I tried that and received the same error. I also tried to delete the detector entirely and cannot save (same error). I tried deleting the phase and saving and also got the same error.