udotdevelopment/ATSPM #47: Version 4.2 Install MOE.Common Error

Chris-Sobie, 2018-12-20:
I was attempting a clean install of Version 4.2 and received the following error when trying to load the website for the first time. Will this same error impact the transition from 4.0 to 4.2? 

[image]

udotdevelopment, 2018-12-20:
It will. I have sent an update to OSADP but it has not yet been updated. Until it is uploaded I have place the correct version of the database conversion tool here on GitHub at https://github.com/udotdevelopment/ATSPM/tree/master/Database%20Conversion%20Executable

amoshouqx, 2018-12-22:
Excuse me. Did you solve this error? Actually I received the same error message as yours when I ran the website. Thank you!

amoshouqx, 2018-12-22:
> It will. I have sent an update to OSADP but it has not yet been updated. Until it is uploaded I have place the correct version of the database conversion tool here on GitHub at https://github.com/udotdevelopment/ATSPM/tree/master/Database%20Conversion%20Executab

Hi, I have met the same problem and I have already downloaded the correct version of the database conversion tool.  However the problem could not be saved. How can I run the website correctly? Thanks!