udotdevelopment/ATSPM #48: Unable to load embedded resource 'MOE.Common.Migrations.DropOrCompressTable.sql'

amoshouqx, 2019-01-03:
Hi I am installing the ATSPM ver 4.2 now and I have just finished the step 2.1. When I was trying to load my local website, the following error rouse(please see the attached picture). I just followed step by step according to the installation manual and had no clue what happened. Could I get some help about how to solve this? Thank you!
[image]

udotdevelopment, 2019-01-07:
Sorry for the delay. I have submitted fixes for this to OSADP back in December but they have not posted them yet. I put them on github so you can access them now. They are at https://github.com/udotdevelopment/ATSPM/tree/master/ATSPM%20Executables

udotdevelopment, 2019-01-07:
I also added a release to GitHub so can get it there as well. https://github.com/udotdevelopment/ATSPM/releases

amoshouqx, 2019-01-08:
> I also added a release to GitHub so can get it there as well. #https://github.com/udotdevelopment/ATSPM/releases

Thanks for the reply! Now we downloaded the updated ATSPM executables and reinstalled the software. Still, we were just finishing step 2.1 and trying to load our website. However, it failed again and the page indicated that the type 'SPM.MvcApplication' could not be loaded. We were suspecting that the Internet Information Services(IIS) manager in our pc has some problem due to the multiple attempts of installing different versions of ATSPM(v 4.0.1, v 4.2, v4.2 with updated ATSPM executables).  So we used another computer for the installation and this same problem arouse when launching the local website. So we wonder what might be the potential reason for this problem? Thanks!
[image]

udotdevelopment, 2019-01-08:
That one is a bit too general of an error for me to be able to tell for sure but I have a few thoughts you might be able to try. First I would make sure to delete the database and relaunch the website. Second I would say it might be your IIS. Make sure that all the correct IIS features are turned on. If you go to udottraffic.utah.gov/atspm under the links menu you will see a "train the trainer" webinar options. One of these we go over which features should be turned on. Your organization may have different requirements for how IIS is set up so it may not be exactly the same.

closed 2019-08-05 by yunongxing