udotdevelopment/ATSPM #49: Set Username and Password Controller

amoshouqx, 2019-01-18:
Hi Dear MDoT:
    I am installing ATSPM and I am on the step 2.8.1 "Install the 'FTPFromAllControllers'". We are using Siemens controllers and we usually retrieve HD data by using WinSCP.  To access the data, we login through WinSCP with a username of "admin" and we do not have a password. Also, we have a private key file in order to connect to a controller.  
    On step 2.8.1.3, in SQL Server Management Studio, I edited the username as "admin" and the password as "none" and then run the executable. However, the connection failed.  I was not sure if I should leave password blank since "none" also looked as a string. So I left the content of "password" blank and tried to connect the controllers again but it still failed.
    So here is my question. Through all the steps, it did not ask me for the path of my "private key file", which is a necessity when we were using WinSCP to get connection to the controllers before. Will this be the reason that the connection could not be successfully made between my localhost and the controllers? If not, what could be the potential reasons? Thanks!

amoshouqx, 2019-01-18:
Thanks for the answering!

closed 2019-01-18 by amoshouqx

gault00, 2019-01-18:
An easier way to use WinSCP would be to create a batch file using the command line interface for WinSCP. You can find the syntax in WinSCP for the session. I’ve used synchronize command to pull the files, and then set up a scheduled task to run the batch file. You could save the .dat files into the folder where the next process looks to convert them to CSV for import to the ATSPM database.