udotdevelopment/ATSPM #51: Step 2.2 Launching the Website

amoshouqx, 2019-03-13:
I had downloaded the ATSPM 4.2 package from OSADP. When I was trying to launch the website, it failed and the web browser indicated that "A network-related or instance-specific error occurred while establishing a connection to SQL Server". I discussed with you in another issue https://github.com/udotdevelopment/ATSPM/issues/44 and did not find a reason for this error.

However, I was suspecting this error appeared due to that some of the files in OSADP package had not been updated yet. So I downloaded the newest release from Github. I deleted the IIS Manager and the website folder (C:\inetpub\wwwroot). Then I reinstall the IIS Manager and used the new ATSPM executable from Github. I thought this might work but it failed again. The browser showed "Parser Error". What could be the potential reasons for this? Thank you!

[image]

udotdevelopment, 2019-03-13:
This looks like an IIS configuration issue to me. I would check that all the correct features are installed with IIS.

Chris-Sobie, 2019-03-19:
I'm having this same issue. I installed IIS on the box according to the user manual.

sbennettcityofsac, 2021-02-10:
I'm having the same problem as well.