udotdevelopment/ATSPM #53: Getting Error doing The Watch dog setting V4.2

MinaNdot, 2019-04-02:
Hi All, 
Doing this setup for Nebraska Dot , I looked on the documentation of WatchDog   and it refrenced that the watchdog email need to be added on the Task config file  , but I figured out that most of new version stuff has moved inside the site instead of the config files, so I logged as an admin and > ADMIN>APPLICATION SETTINGS > WATCHDOG setting and I get the below error :

Compilation Error 
  Description: An error occurred during the compilation of a resource required to service this request. Please review the following specific error details and modify your source code appropriately. 

 Compiler Error Message: CS1061: 'MOE.Common.Models.WatchDogApplicationSettings' does not contain a definition for 'EmailAllErrors' and no extension method 'EmailAllErrors' accepting a first argument of type 'MOE.Common.Models.WatchDogApplicationSettings' could be found (are you missing a using directive or an assembly reference?)
 
--
Line 104: Line 105:    <div class="form-group"> Line 106:        @Html.LabelFor(model => model.EmailAllErrors, htmlAttributes: new { @class = "control-label col-md-2" }) Line 107:        <div class="col-md-10"> Line 108:            <div class="checkbox">

c:\windows\system32\inetsrv> "C:\Windows\Microsoft.NET\Framework64\v4.0.30319\csc.exe" /t:library /utf8output /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\ea2fc76a\00d87768_e793d401\AlexPilotti.FTPS.Client.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\d9ed4a4a\00d87768_e793d401\Microsoft.AspNet.Identity.EntityFramework.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_64\System.EnterpriseServices\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.EnterpriseServices.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\Microsoft.Web.Infrastructure\v4.0_1.0.0.0__31bf3856ad364e35\Microsoft.Web.Infrastructure.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\819ce9d0\00d87768_e793d401\System.Web.Optimization.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\5ed51041\00d87768_e793d401\NuGet.Core.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\81f0a545\00d87768_e793d401\WebGrease.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\00e10da7\00d87768_e793d401\CsvHelper.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\d8b99f2e\00d87768_e793d401\EntityFramework.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\Microsoft.CSharp\v4.0_4.0.0.0__b03f5f7f11d50a3a\Microsoft.CSharp.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\66126a07\00d87768_e793d401\Caseiro.Mvc.PagedList.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Xml\v4.0_4.0.0.0__b77a5c561934e089\System.Xml.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.ServiceModel\v4.0_4.0.0.0__b77a5c561934e089\System.ServiceModel.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Web.Extensions\v4.0_4.0.0.0__31bf3856ad364e35\System.Web.Extensions.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\a304e9f0\00d87768_e793d401\Microsoft.Extensions.Caching.Memory.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\0dc371e4\00d87768_e793d401\Microsoft.Threading.Tasks.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.ServiceModel.Web\v4.0_4.0.0.0__31bf3856ad364e35\System.ServiceModel.Web.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\e22537b5\00d87768_e793d401\Microsoft.Owin.Host.SystemWeb.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\9edfb80b\00d87768_e793d401\System.Web.Razor.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Web.ApplicationServices\v4.0_4.0.0.0__31bf3856ad364e35\System.Web.ApplicationServices.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\mscorlib.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\b9fd2fc5\00d87768_e793d401\Microsoft.VisualStudio.TestPlatform.MSTest.TestAdapter.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.WorkflowServices\v4.0_4.0.0.0__31bf3856ad364e35\System.WorkflowServices.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Drawing\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Drawing.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\App_global.asax.7yzrhx9m.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\6cfa69dc\00d87768_e793d401\Microsoft.Owin.Security.MicrosoftAccount.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Web.Services\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Web.Services.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\98f6bfc0\00d87768_e793d401\System.Web.Mvc.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\85149c70\00d87768_e793d401\Microsoft.Owin.Security.Cookies.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\d0e7dfb4\00d87768_e793d401\Microsoft.Extensions.DependencyInjection.Abstractions.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\6ba43bfb\00d87768_e793d401\Microsoft.Extensions.Logging.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\30ef70fa\00d87768_e793d401\MvcCheckBoxList.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Configuration\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Configuration.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\19651cec\00d87768_e793d401\Microsoft.Owin.Security.OAuth.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_MSIL\System.Runtime\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Runtime.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\3d58f2ec\00d87768_e793d401\Microsoft.Extensions.Primitives.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\9a9431b8\00d87768_e793d401\System.Web.WebPages.Razor.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\1fb2e6b0\00d87768_e793d401\Microsoft.Extensions.Options.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\7b441dfc\00d87768_e793d401\SharpSnmpLib.Portable.dll" /R:"C:\Windows\Microsoft.Net\assembly\GAC_64\System.Web\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Web.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\9ce75c75\00d87768_e793d401\Microsoft.Owin.Security.Google.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\f04fff1e\00d87768_e793d401\Microsoft.Extensions.Logging.Abstractions.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\788d72da\00d87768_e793d401\Microsoft.Extensions.Caching.Abstractions.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\af044a57\00d87768_e793d401\Microsoft.Threading.Tasks.Extensions.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\e633cc74\00d87768_e793d401\System.Web.Helpers.dll" /R:"C:\Windows\Microsoft.NET\Framework64\v4.0.30319\Temporary ASP.NET Files\atspmqa\8fa7fa7d\fa060d91\assembly\dl3\b62fd42a\00d87768_e793d401\Microsoft.Owin.dll" /R:"C:\Windows\Micros [... cut ...] contain a definition for 'EmailAllErrors' and no extension method 'EmailAllErrors' accepting a first argument of type 'MOE.Common.Models.WatchDogApplicationSettings' could be found (are you missing a using directive or an assembly reference?) i:\NDOTQA02\ATSPMQA\Views\WatchDogApplicationSettings\Edit.cshtml(110,59): error CS1061: 'MOE.Common.Models.WatchDogApplicationSettings' does not contain a definition for 'EmailAllErrors'
--

Can someone assist me with that, not sure what I am messing here.

udotdevelopment, 2019-04-03:
There is a value in the database in the settings table that you need to set if it did not get set during installation (possibly a bug). It is in the ApplicationSettings table. If the value for ID 1 is null it needs to be set to false.

MinaNdot, 2019-04-03:
@udotdevelopment the Value of which column? if you mean EMAILALLERROS it seems there is no column with this name, because I looked at the code here it seems you drop the table after you create it:

public override void Up()
--
9 | {
10 | AddColumn("dbo.ApplicationSettings", "EmailAllErrors", c => c.Boolean());
… |  
15 | DropColumn("dbo.ApplicationSettings", "EmailAllErrors");
16 | }
17 | }
18 | }

udotdevelopment, 2019-04-03:
Yes it should be column EmailAllErrors. What you have listed previously are 2 separate functions that do not both run. If your column is not present then your database did not set up correctly. Is this a new install or an update? Also please check the _MigrationHistory Table and tell me what the last migrationId in the table is.

MinaNdot, 2019-04-03:
There is no Column with this Name in the dbo.applicationsettings so I created one with Bit value and give it a false for ID 1 still getting the same error.
This is totally a new installation and I used the 4.2 web files , here is the MigrationHistory last value : 
MigrationId	ContextKey	Model	ProductVersion
201808171744257_AggreationLongsForId	MOE.Common.Migrations.Configuration	0x1F8B0800000000000400ED7D5973DCB8B2E6FB44CC7F50E8E92E7D2DDB7DFA4C9F0EFBDE90B5B41D2D59D592DC3DFDA4A049A8C4308BAC43B2B4C4C4FCB279989F347F6100AE581220369255B25E6C15012412890F892D91F9FFFECFFF7DF75F8FAB64EF1EE5459CA5EFF7DFBC7ABDBF87D2308BE274F97E7F53DEFEC7CFFBFFF59FFFFDBFBD3B89568F7B7FB4F97E24F970C9B478BF7F5796EB5F0E0E8AF00EAD82E2D52A0EF3ACC86ECB5761B63A08A2ECE0EDEBD7FF3878F3E6006112FB98D6DEDEBBCB4D5AC62B54FDC03F8FB23444EB721324E7598492A2F98E53AE2AAA7B9F83152AD64188DEEF9F5F9CBC3ACA56AB2C7D5567DEDF3B4CE20033728592DBFDBD204DB33228319BBF7C29D0559967E9F26A8D3F04C9F5D31AE17CB74152A086FD5FFAECBA2D79FD96B4E4A02FD8920A374599AD0C09BEF9B111CD015FDC4AC0FB9DE8B0F04EB090CB27D2EA4A80EFF70F4342F82C5BEEEFF1B5FD7294E4242720E0575DB11FF6FAC41F3A34BC7DF5E6EFAF5EBF7AFDC3DED1262937397A9FA24D9907C90F7B8BCDD7240E7F434FD7D93794BE4F374942738879C469CC07FC6991676B94974F97E896E7FBD3F1FEDE014BE080A7D095870AD74DFC94963FBEDDDFFB8CD909BE26A88304258EAB32CBD1AF28457950A268119425CA718F7E8A502554810DAED2635CAAAD8DFC7D8DD10E54A82672B8C443F169986D3515D25D98E996081E11786CEFEF9D078F67285D9677EFF7DFFEF4D3FEDE69FC88A2F64B43F84B1A6355800B95F966B09EAB78990649CF2D5011FE53AB22C316927F15B5BE79FDDA4BAD9F83FB7859A103441856449728A9D28BBB785DEBA37EECDC74B94EF36C7599253434DBC49BAB6C9387A431992CC775902F5169C05C05A221DE9A4C206B559A8AB33A832963E708775558E9E401EE989C108B5406059F742E88D97707BDC2D450A3363A745E056AAF3D27559DA808F3785D8B583AA87F9A644C63D87819D62DDCE4C3BA85AD1D209BE16B04C8AACC6C80ECA63663406A4E8A5B0A48ABFEED3597591FF7E5E6EAE79A039B7EEE4BBAF6F3E72C1D5E24DD057939B08E1869F572F8F56B8EEEE360005F23D57E75973D5CA47FA2AF45DCAF583F64584B05A925B1C3E53247B552BDB221EAAE938D160C72DD0C2D2B74796D56DD30A735E1268B8C5B69268163794E53AE8F5189AAE6CB97654C1619EFD24C02EFF29C4E332223121BA5D9149D4B6F36D5DB284EAAE85433642317D79DAAC71DA4BA22B21BBF2A83D5DA798BDE08FB1A3DAA36D8BE74B7F55ECA51E1F07BAA61D5A4CB78DDE520CF75D24DAF487B4EB92441A9F0E94EAAA465D14487D465E6521ED47034551EDA23D99FF2D018F56FFC6CF6E0B6363BDCC84D777DCECA19D68E5889E5A5B3125BE4F12AC89F9C8FD19ADFA66D40619646F3D5FF69711845392A0A97CD9D5DD5677800959B48D5EA1F47AA394B9733557D89961ED60A47595AE65992A04A07BB523B4949E6C8E35665BDCEB320BC939D6D36C937EDE4426D52D824717BC2A57BDD9818CCB9FC9A403627EBF34577A89A3B3627C0239D41CE2993CB94DF1AC62A3EDB1C027F75828CAF26D5941F663653B1C56514B863D2654CB299DC4ED31B401B1E5F36A5663BC06CEAB73AC2A4CA6ED12ACBCFDECAD35EF038CEFB730167625A27B87EDA7FBEF8C8313BB4F8CAC80908EED2BBA0409F37ABAF28776BEDA782A579718FF224581B4F6E3CA3285FC54511DF2339A7838CB1346C391B38BFCAF2815997CA064CBC5DAA789728663155D30CAE07B864B3429CD23914DC32D93C6ED10D963152E6B8658ED514D27687D914D2969A6B0AB1993AA6BDF7AAE5A39C343CDDC38A551FDDE196A0C4751629CA200D11C12596C6FA4360A6B1CEE3F46A8D50741A27A5A1B223BB71BCB1EC3715D6DB7352103784D9A1F4C4867698A98DA63ECFEE5175AE094DBEC3355A143B46614CC6DD228B7B5B25235E8FB17E7A3264B52477C94759DEE8C74EBA191EE1163779DAEB3A9D8187F9F918E4D143909BAF7F06F7A57EE647E9DE549C428D6FA0DAB62B2EA1B2FC06C8CEDF4241B984E94891D57886D7BD43A36B93DCA009593418B73B886FE9288F07B8CAE8DC52CEFB4C43BC53394DB96F958E9AEB3E17C06D9B28E7B2CB616CE946695335876C4E804B3A839C53269787951533BECC975854F1EF6DADA5637B3285D98979879B5B1B3145E7EA6886099B5E17084C0101BF0714236D9F1D662DD8EA433EBBF9BAF77634568115ACCAACC569FB6A65AEC2157E3158D1D9DABAAED09F9D01497F82E27FD93730F88105A2D528EA577726C3A72D35D7B8A1B7CBA60347BAD59E75961ABC361ED34A56EF21939FA512BB58373231A44ACE6699CD1DF118DB67AB8E885E00E8F738342ED649F0749147667743FA2B10F6AAC168FD41179D6DD5CFDF161AAFFACDAE1BB710CE13A27766B0F2F61F266865CBCEB758E68D97CCD7CC66E64F530256D7586EC060E1F3F96291E5F45DC0DFFF365ACB4EAF17B50AC8F227A775B8DE5C8255CD3DC275BADEC97F29503E707CE4A7371641513C6479E4B922ED31DFDA52998CF5BACCCB89E368E3D8E14523678466D2AD4CD197DE9DD2DAC9C4A60F4BB96AC9C9BDF191165FFA7BEB64DD532C72E054783985A2243EC56DC4DC07E0C20B847B9463799FE1FF3B8393DA64964E29CC0FF79240F9ACC08FC5E1E9260DB764B05B8FF397210ED7E569341A9887B43D7285CA125727B5A4E4F3DD303060CC465459211B12657E37471950FB2C21DB1278812E5C17252AAFF64B3DC8C6052660B1AA0564BB03B8A00CBE06053ACCC33BBC35B403A784881780EE7DC084A999111A494A38D44FAC308B1487CEFE26505299B95F933295AB40C44CDFCCF7212BC2B4BC2BAEB3DF105A7F4A23F4686883D89726BD6054B811061E6077A3AF155A891DE3FF4B749193537C46646CC2C0E45419965EDC5E660F85518BAB67ABF542D5A0142970BCC999334CA0A0F698AB555B6237D6B8C2DB31C63EAD8225FA9227A3C3A8AA6812C05EA2A3605D86774127B0E9AABC42618ECA49AA0C1E88DA38CA36697916AF629505B436BEFF0C70238EB3A5F3A24741683B707F84D70028DC9043D54A846EC61DE7715ABD4EBA268F95D27A85E1467181F04A222DAFEF7254DC65496FC56F67678E31778C96394217B78B004B3541495CAC1C7DAF84417A1C3C559AF9235EF978A1769246EEB416E4E22ADB1498DEE27C81826F8CD3034F3431A7CEA089579BD5250AB33C7284CB9F087D8B82A78B3479725D211DA3DB000FC193551027C36E0E3C9D47E025F6A4159E650F1F63717459BA02209C5FA1FCBEBFBD1CEFAD68F04840B340112A701D417DDC69841EE367D6474F618228176D76CFAE792AB36D82238B4D308D10CB3B45DD4DF087381516BB2EA7B4F5FB2247885FE21D13CD8FDD2CF4174A92ECC103A15FF14C967AA0738DFB2CA980E93C5D478761B9F131F3F36FB1FD7B1851BEE4E287E94D5F4A7CD825CD2C7DBC2B2FE1C52344A50DAB059DB3BE0249BD28AD1D525A7D0FCE39228D31BC082367F0B2345E50BB43A83DCCF3F81E67BC48AB59CE17B14BE48DAF7A1A9F7996E3 [... cut ...] 18EF4DF6390B102732A468E79AA44B7B77505B20361FF04FBC2A0B96E83C8B5052545FDF1D5C6E70E915AA7F1DA3028FAC8EC43B4C33AD154B4FB4CDF329BDCD5A8F2C1C476D9636B9B74E0AA2A00C0EF332BE0DC2B231848ED3E5FEDE1F41B2C1594E565F51F429BDD894EB4D899B8C565F13C60A8D787651D5FFEE40E0F9DD45B5EE2A7C3401B319E326A08BF4C3264EA28EEF53AC00B94E9391202E637E45F87BDD9725FE1F2D9F3A4A9FB354935023BECED3CD355E6026985871915E05F7C886370CBF33BC180B9FF0F7FB382288961119EE0856ECEF8EE3609907ABA2A1D197C73F3186A3D5E37FFE7FB384434AB94E0300	6.1.3-40302

MinaNdot, 2019-04-03:
Just to mention I installed this application on Server 2016 with probably .net 4.7 will this be the Problem ?

udotdevelopment, 2019-04-03:
It should not be a problem. Your last migration is missing. Adding the column should have addressed the issue but would cause future updates to fail. I will do a test on my end and see if I can reproduce the error.

MinaNdot, 2019-04-03:
@udotdevelopment Just ran a profiler on the database, seems the code is not sending anything to look for this column, but I noticed it runs this command 
exec sp_executesql N'SELECT COLUMN_NAME FROM INFORMATION_SCHEMA.COLUMNS where TABLE_NAME=@Table',N'@Table nvarchar(11)',@Table=N'AspNetUsers'
Can it be looking for Emailallerrors In this table ?

udotdevelopment, 2019-04-03:
No. I think I know where the error is occuring but I will need to make the fix test it and then upload it to github. I will let you know when it is available.

MinaNdot, 2019-04-03:
@udotdevelopment last thing the select command send by the application is not Including EmailAllErrors:

SELECT 
    [Limit1].[C1] AS [C1], 
    [Limit1].[ID] AS [ID], 
    [Limit1].[ApplicationID] AS [ApplicationID], 
    [Limit1].[ConsecutiveCount] AS [ConsecutiveCount], 
    [Limit1].[MinPhaseTerminations] AS [MinPhaseTerminations], 
    [Limit1].[PercentThreshold] AS [PercentThreshold], 
    [Limit1].[MaxDegreeOfParallelism] AS [MaxDegreeOfParallelism], 
    [Limit1].[ScanDayStartHour] AS [ScanDayStartHour], 
    [Limit1].[ScanDayEndHour] AS [ScanDayEndHour], 
    [Limit1].[PreviousDayPMPeakStart] AS [PreviousDayPMPeakStart], 
    [Limit1].[PreviousDayPMPeakEnd] AS [PreviousDayPMPeakEnd], 
    [Limit1].[MinimumRecords] AS [MinimumRecords], 
    [Limit1].[WeekdayOnly] AS [WeekdayOnly], 
    [Limit1].[DefaultEmailAddress] AS [DefaultEmailAddress], 
    [Limit1].[FromEmailAddress] AS [FromEmailAddress], 
    [Limit1].[LowHitThreshold] AS [LowHitThreshold], 
    [Limit1].[EmailServer] AS [EmailServer], 
    [Limit1].[MaximumPedestrianEvents] AS [MaximumPedestrianEvents]
    FROM ( SELECT TOP (1) 
        [Extent1].[ID] AS [ID], 
        [Extent1].[ApplicationID] AS [ApplicationID], 
        [Extent1].[ConsecutiveCount] AS [ConsecutiveCount], 
        [Extent1].[MinPhaseTerminations] AS [MinPhaseTerminations], 
        [Extent1].[PercentThreshold] AS [PercentThreshold], 
        [Extent1].[MaxDegreeOfParallelism] AS [MaxDegreeOfParallelism], 
        [Extent1].[ScanDayStartHour] AS [ScanDayStartHour], 
        [Extent1].[ScanDayEndHour] AS [ScanDayEndHour], 
        [Extent1].[PreviousDayPMPeakStart] AS [PreviousDayPMPeakStart], 
        [Extent1].[PreviousDayPMPeakEnd] AS [PreviousDayPMPeakEnd], 
        [Extent1].[MinimumRecords] AS [MinimumRecords], 
        [Extent1].[WeekdayOnly] AS [WeekdayOnly], 
        [Extent1].[DefaultEmailAddress] AS [DefaultEmailAddress], 
        [Extent1].[FromEmailAddress] AS [FromEmailAddress], 
        [Extent1].[LowHitThreshold] AS [LowHitThreshold], 
        [Extent1].[EmailServer] AS [EmailServer], 
        [Extent1].[MaximumPedestrianEvents] AS [MaximumPedestrianEvents], 
        '0X0X' AS [C1]
        FROM [dbo].[ApplicationSettings] AS [Extent1]
        WHERE [Extent1].[Discriminator] = N'WatchDogApplicationSettings'
    )  AS [Limit1]

yunongxing, 2019-04-04:
Hi, I just posted a fixed version for this issue, v4.2.0.1, please update yours.

MinaNdot, 2019-04-04:
Thanks, I am out on a conference today and will test tomorrow and let you know

Mina

Sent from my Verizon, Samsung Galaxy smartphone

MinaNdot, 2019-04-05:
Thanks , I can confirm that this update works as expected.
I only had to set the Emailallerrors to False as by default it is set to Null, as we talked on the other Bug.

yunongxing, 2019-04-08:
From Mina: "Thanks , I can confirm that this update works as expected.
I only had to set the Emailallerrors to False as by default it is set to Null, as we talked on the other Bug."

closed 2019-04-08 by yunongxing

ZachParkerCR, 2019-07-03:
I know this is closed, but I am also having this same issue and was wondering what to do with the fix that was released?  I'm not sure what file to grab and where to place it on my ATSPM 4.2 server.

Thanks,
Zach

MinaNdot, 2019-07-03:
You have to upgrade the whole site to  version to 4.2.1, Grab the release from the download section and replace all the files in the web folder and also if you have Tasks running you have to upgrade all of them and replace their files.

Mina Nashed | Senior IT Infrastructure Analyst
ocio
Nebraska Office of the Chief Information Officer
OFFICE: 402-471-1181   |   CELL: 951-426-7088

ZachParkerCR, 2019-07-03:
Can you point me to where the download section is?  I'm not sure where that is in GitHub.  

Also, should I just wait for version 4.2.2.  I heard that's about to be released any day now.

MinaNdot, 2019-07-03:
https://github.com/udotdevelopment/ATSPM/releases   download 4.2.0.1 ATSPM.Executables.zip
Mina Nashed | Senior IT Infrastructure Analyst
ocio
Nebraska Office of the Chief Information Officer
OFFICE: 402-471-1181   |   CELL: 951-426-7088

ZachParkerCR, 2019-07-03:
Mina, thank you for your quick replies.  

This is my first time attempting an upgrade from one version to another.  Are there steps I can follow to go from 4.2 to 4.2.0.1?  I just got the installation setup on 4.2 yesterday and don't know the procedure to go to a newer version.

MinaNdot, 2019-07-03:
I was like you one month ago ☺ , Just replace the files not too much steps. In state of Nebraska we have Beyond compare application that checks for files changes and I use that to copy different or new files then restart the app pool after that, make sure to change the database and replace the null value as this is still a different bug not fixed, review my comments above.

Mina Nashed | Senior IT Infrastructure Analyst
ocio
Nebraska Office of the Chief Information Officer
OFFICE: 402-471-1181   |   CELL: 951-426-7088

ZachParkerCR, 2019-07-03:
It's been a "fun" process so far interpreting the instructions.  :)  I'm just wondering which files to replace - all the files in all the executables folders, or just the ATSPMWebpage folder's files?  I'm assuming I'll need to re-edit the Web.Config files to adjust my connection strings and all the other customization settings done, because those files will be overwritten.

MinaNdot, 2019-07-03:
Yes , all the file in webpage folder, but if you setup the Task you will need to replace their files accordingly.
I didn’t replace the webconfig as far as I remember, but you can replace it and copy the connection strings over yes.
Usually in any updates I keep the webconfig unreplaced unless something was added there, I am talking generally in any web app.

Mina Nashed | Senior IT Infrastructure Analyst
ocio
Nebraska Office of the Chief Information Officer
OFFICE: 402-471-1181   |   CELL: 951-426-7088