udotdevelopment/ATSPM #59: Split Monitor Showing Partial Data

MinaNdot, 2019-06-05:
We are facing an issue with Split Monitor on 4.2.0.1 Version, Some Signals are showing Partial data for some phases Like Phase 1 and 5 Please check attached screenshots.

As per the Admin of the site the data is in the database and he verified that but not sure why it miss these time frame, any advice will be helpful
[image]
[image]

.

MinaNdot, 2019-06-06:
Attaching an explanation file from the user. 
[ATSPM_Screenshots.docx](https://github.com/udotdevelopment/ATSPM/files/3261774/ATSPM_Screenshots.docx)

taylormd3, 2019-06-06:
What controller brand & version number is being used?  Also, how often are you retrieving data?

taylormd3, 2019-06-06:
> Attaching an explanation file from the user.
> [ATSPM_Screenshots.docx](https://github.com/udotdevelopment/ATSPM/files/3261774/ATSPM_Screenshots.docx)

What controller brand @ version number is being used? Also, how often are you retrieving data?

MinaNdot, 2019-06-06:
> What controller brand & version number is being used?  Also, how often are you retrieving data?

2070 controllers with Intelight MaxTime local software v. 1.9.14 and 2.0.16. 
We running the Task every 15 minute.

taylormd3, 2019-06-06:
The problems is most likely that you're not running the task often enough and that the data is getting overridden. Intelight currently only buffers 4000 lines of code before it starts to override. For a normal intersection, this is only about 8 minutes. The task needs to be changed to every 5 minutes or more often. Intelight says they're working on changing this to be 4x longer on the buffering, but that is most likely the issue is that you're missing data.

MinaNdot, 2019-06-06:
But Mark we are missing a whole periods of time, and Phase 5 in some cases are completely not reporting any records.
I can try your suggestion but have you looked on the database screenshot, seems as per the user the data is there.

Mina Nashed | Senior IT Infrastructure Analyst
ocio
Nebraska Department of Health and Human Services
OFFICE: 402-471-1181   |   CELL: 951-426-7088
DHHS.ne.gov<http://dhhs.ne.gov/Pages/default.aspx>  |  Facebook<https://www.facebook.com/NEDHHS/>  |  Twitter<https://twitter.com/NEDHHS>  |  LinkedIn<https://www.linkedin.com/company/nebraska-department-of-health-and-human-services?trk=company_logo>

taylormd3, 2019-06-06:
You bring up a good point - but sometimes when data gets overridden it may do weird things.  I would try to first change the retrieval rate to 5 minutes or less to see if that fixes the problem.  If not, i would reach out to Intelight (Shane Johnson) and ask him for guidance as he is in a unique position working for Intelight but also wrote the metric when he worked for UDOT.

mdbowie, 2019-07-18:
Mina Nashed.
We ran into a similar problem at the City of Mississauga which uses the Intelight controller. Tracked the problem down to the lack of an all-red interval on phases 1, 3, 5 and 7. We resolved the problem by adding a 0.1 second in the all-red.

Hope this helps.

Mike

MinaNdot, 2019-07-18:
Thanks Mike, we will give a shot.
Appreciate your assistance.

Mina Nashed | Senior IT Infrastructure Analyst
ocio
Nebraska Office of the Chief Information Officer
OFFICE: 402-471-1181   |   CELL: 951-426-7088

closed 2019-08-15 by udotdevelopment