udotdevelopment/ATSPM #63: SQL SELECT/EXECUTE errors on website launch

dbrandesky, 2019-07-15:
Hi all, I've been following the GDOT instructions and installing ATSPM 4.2. So far, the only major change I had to make was to specify .\SQLEXPRESS instead of localhost as the database to which the IIS app will connect. However now I'm stuck with the following errors appearing in my web browser:
`The SELECT permission was denied on the object 'syscategories', database 'msdb', schema 'dbo'.
The SELECT permission was denied on the object 'sysjobs', database 'msdb', schema 'dbo'.
The EXECUTE permission was denied on the object 'sp_add_job', database 'msdb', schema 'dbo'.`

I created the 'atspm' user in SQL management studio exactly as the GDOT instructions specified, but it seems like this user needs some rights in the msdb database as well. I've tried various things but it all seems to go in a circle: I fix one problem and create two more. Any suggestions or tips on how to get out of this problem?

udotdevelopment, 2019-07-15:
You need elevated permissions. It looks like you need the admin permissions so you can create databases.

dbrandesky, 2019-07-15:
I just tried making 'atspm' serveradmin and sysadmin, now I have another error:

`The specified '@owner_login_name' is invalid (valid values are returned by sp_helplogins [excluding Windows NT groups])`

I'm not sure where to find what login name it was trying to use, but I would assume 'atspm'?

udotdevelopment, 2019-07-15:
The login name should be the ID you set in the connection string.

dbrandesky, 2019-07-15:
Interesting, the login name in my connection string is 'atspm', the same user I created. I'm basically following the GDOT install manual word for word.

dbrandesky, 2019-07-15:
Just for funzies, I tried enabling the built-in 'sa' account in MSSQL, then set the connection string in IIS to use the 'sa' account. Refreshed my browser, and I get the exact same error as before:

`The specified '@owner_login_name' is invalid (valid values are returned by sp_helplogins [excluding Windows NT groups])`

udotdevelopment, 2019-07-16:
Something to try is to log in through sql server manager with the user you set up. Check to see if you can access your database and tables. Also follow the directions on the error and execute sp_helplogins.

dbrandesky, 2019-07-16:
I can definitely log in via SQL management studio using the login I created, and it appears I can at least view both the MOE and msdb databases as that user. I ran the sp_helplogins query but all it did was show all the valid SQL logins, and the login I created was listed, so it is valid.

dbrandesky, 2019-07-17:
Fixed this by creating a user in SQL called 'SPM', with the same dbcreator and public permissions as the 'atspm' user I created from the GDOT manual. Issue #52 indicates that there is a hard-coded reference to the 'SPM' username in some module.

closed 2019-07-17 by dbrandesky