udotdevelopment/ATSPM #65: Why does FTPFromAllControllers.exe disable then re-enable data logging on the ASC/3 after FTP upload?

dbrandesky, 2019-07-19:
I was doing some testing with my ATSPM system, and ran FTPFromAllControllers.exe from the command prompt to see what it dumped. I noticed at the end of all the FTP uploads, it attempts to SET object ID 1.3.6.1.4.1.1206.3.5.2.9.17.1.0 to integer 0, then SET it back to integer 1. This object ID controls the hi-res data logging on the controller; 0 disables, and 1 enables. Why was it deemed necessary to disable, then re-enable logging each time you do an FTP upload?

Also note that on my system, this step always fails, but I have a feeling it's because I'm uploading logs over an SSL VPN connection to get to my test controllers, so UDP is not so reliable.

yunongxing, 2019-08-05:
The comment in FTPfromAllControllers\Program.cs says:
                //Turn Logging off.
                //The ASC3 controller stoploggin if the current file is removed.  to make sure logging continues, we must turn the loggin feature off on the 
                //controller, then turn it back on.

So it's definitely intended behavior.

closed 2019-08-15 by udotdevelopment

shanejohnson-atspm, 2019-08-15:
If we remove the active log from and ASC# controller, logging will stop until either a new log file is created by the controller at the start of the next 15 minute interval, or the controller is told to stop and then start logging.

The newest version of the program gives you the option to leave the active log file, and does not require the logging feature to be restarted.

dbrandesky, 2019-08-15:
Ah, I get it now. Traditionally ASC/3 used a .dat file for MOE data, so there was no way to distinguish the file that was still open for writing on the controller from the ones that were ready to go. As of ASC/3 2.66, the controller now compresses the dat file into a .datZ file, so ATSPM would only want to upload the .datZ files, not the .dat files.