udotdevelopment/ATSPM #74: Create chart activation error

abdullah-traboulsi, 2019-11-19:
After installing ATSPM 4.2 downloaded from OSADP website and defining MOEWcFService, I have tried to create a chart such as Split Monitor. But it threw an error asking for service activation.

error statement

> The requested service, 'http://localhost/MOEWcfService/MOEWcfServiceLibrary.MetricGenerator.svc' could not be activated. See the server's diagnostic trace logs for more information.

udotdevelopment, 2019-11-19:
Have you verified the wcf service is running correctly by browsing to it in
iis? If you can browse to it copy the address provided in the link and put
it in your configuration.

abdullah-traboulsi, 2019-11-19:
Following Installation manual, step 2.7 "Create Web Services", I have created and installed the MOEWcfService application in IIS. 
I am able to browse to these services, but it shows an error:

> The Web server is configured to not list the contents of this directory.
http://localhost/MOEWcfService/

Additionally, I have modified the web.config for the ATSPM main website to point to these new services

udotdevelopment, 2019-11-19:
You can't browse directly to the application in IIS. In IIS click on the
WCF service application and click on the content view tab. Right click on
the .svc file and click browse. This should show you if the service is set
up correctly.

abdullah-traboulsi, 2019-11-19:
After checking the service as you said, it showed an error which indicates that the service wasn't set properly. I have modified my web.config "Key" value to point to "http://localhost/ATSPMImages/", and now the service is registered and working correctly.

Now, I have tried to generate a "Split Monitor", but nothing is shown. It is blank (instead of the previous activation error).

udotdevelopment, 2019-11-19:
If there are no errors showing than the image was generated. Make sure that
you have set the Image Location under Admin -> General Settings.

abdullah-traboulsi, 2019-11-19:
I have updated the Image Location under Admin -> General Settings, but still the same not showing any images.

Btw, I have checked the physical folder for ATSPMImages, and there are no images there to view.

udotdevelopment, 2019-11-19:
Is your Image Path set under general settings? Also make sure there is a
trailing slash on each path. Another thing to check is that your wcf
service has access to the physical folder.

abdullah-traboulsi, 2019-11-19:
I have set both Image Path and Image URL to  "http://localhost/ATSPMImages/" with the trailing slash.

As for checking if WCF service has access to the physical folder, I have added the security privileges to read permission to  IUSR and IIS_IUSRS.

Unless you mean differently?

udotdevelopment, 2019-11-19:
That is what I mean by the trailing slashes but Image path has to be a
physical folder not a URL.

abdullah-traboulsi, 2019-11-20:
I have set the path to the physical folder, but still, no chart is generated.

FYI, right now, my application is not connected with any controller yet to retrieve data from intersections.
How can I check which code is executed to retrieve data from SQL to print in the Split Monitor chart?

Regarding WCFService, when browsing it using IIS, it shows the following:

> You have created a service.
>To test this service, you will need to create a client and use it to call the service. You can do this using the svcutil.exe tool from the command line with the following syntax:
>svcutil.exe http://localhost/MOEWcfService/MOEWcfServiceLibrary.LinkPivotService.svc?wsdl
>You can also access the service description as a single file:
>http://localhost/MOEWcfService/MOEWcfServiceLibrary.LinkPivotService.svc?singleWsdl
>This will generate a configuration file and a code file that contains the client class. Add the two files to your client application and use the generated client class to call the Service.

abdullah-traboulsi, 2019-11-21:
If there is no information from controllers yet, would the Split Monitor show an empty image? Or it wouldn't show anything?

How can I test my app with some dummy data to populate the Split Monitor chart?