udotdevelopment/ATSPM #83: Trouble Getting Speed Data from Wavetronix Advance Devices

taylorbreannaray, 2020-07-21:
My team and I have been continually troubleshooting and trying to get the SpeedListener component to work properly so we can actually receive speed data that is being pushed from our local Wavetronix SmartSensor Advance devices.

The service is running constantly on the machine as it should, and we would expect to see the data dumped into the proper **dbo.Speed_ Events** table, but nothing at all is there. We have also spoken with a representative from Wavetronix to ensure that the devices are configured as they should be throughout our local intersections.

Does anyone have any insight as to what some issues might be? Or at least be able to guide us in a particular direction?

Also, I noticed that the UDPSpeedListener.cs and TCPSpeedListener.cs files specify port 10088 for listening, does this port also need to be specified in the Admin section in the browser for the Click 650? Or should this be some entirely different port?

The area for specifying said port is shown in the screenshot in red below:

(Note: The area for entering the IP address has been annotated and marked out for security reasons; the port also shown was only being utilized for testing purposes).

[image]

AndreRSanchez, 2020-07-21:
> @udotdevelopment <https://github.com/udotdevelopment>
> @AndreRSanchez <https://github.com/AndreRSanchez>

I will get back with you about this.

taylormd3, 2020-07-22:
I look forward to Andre's comment regarding some of the technical issues with the server but let me explain how to set it up in the Wavetronix smartsensor manager and Click 650/656 device. 

1) SSM Advance: Go to: Channels-Alerts-Zones. Go to: Setup Channels-Alerts-Zones. Go to: Channel 2, enable it and ensure it's enabled. The type should be "Pulse" and a small zone (10 feet long) should be located somewhere between 350 to 500 feet from the stopbar.

2) SSM Advance: Go to: Channels-Alerts-Zones. go to: Setup Output Communication. Ensure that the "Select port to transmit output:" is set to port 2. Ensure that the "Select trigger speed options:" is set to Port 2, channel 2. Our tag is a 6 digit number and needs to be unique. The first 4 digits is the signal id number and the last two digits is the channel number of the output in the signal cabinet. 

3) in the browser, type in the IP address of the click 650/656. Go to "Admin". Under the "Trigger Speed Message Forwarding", please check the "enable forwarding" box. The "append timestamp" box should be unchecked.  Please be sure to enter a correct address with the port being 10088. 

Assuming the setup is correct as explained in items 1-3, Andre can further explain the networking side on the server.  Thanks

Mark Taylor

taylorbreannaray, 2020-07-22:
Mark,

Your feedback is appreciated greatly.

I'm not sure that we currently satisfy all of the conditions you've described above, so I will go back in and make sure that the configurations for our Advance devices do. This certainly points us in the right direction and I hope something good comes of it.

Cheers!

taylorbreannaray, 2020-07-24:
@AndreRSanchez Do you have any other tips for our team? Even after following all of Mark's steps, we still find that dbo.Speed_Events is not being populated.

I do have a question that may help you to help us. We have to connect to our local agency's VPN network before we can make any configurations to our Advances. With that being said, given the fact that we are constantly within this VPN network, is that affecting the IP that we should be using on the Click 650 Admin settings?

shanejohnson-atspm, 2020-07-24:
The data path is, more or less, like this:

Advance Sensor->Click6xx->Speed Listener Service->Database Table.

Your firewall rules must allow inbound traffic from the Click6xx.  The speed listener service must be running using using the port you configured in the Click6xx.  The service must be able to write to the database using the values in the config file of the service.

taylorbreannaray, 2020-07-31:
All,

After taking all of the comments into consideration, I was able to get it figured out and working.

Thanks so much!

closed 2020-07-31 by taylorbreannaray