udotdevelopment/ATSPM #93: Timeout Exception during FTP 4.2.4

NDSU-UGPTI-ATAC, 2021-02-22:
We are in the process of installing v4.2.4 and when running ftpfromasc3-ftpfromallcontrollers for Econolite controllers it keeps erring out with a TimeoutException. We are able to ping the controllers. What we have tried for troubleshooting is changing port number from 161 to 22 (for sftp) and increasing timeout duration. Anyone else run into the same issue? 

The same program worked for the most part in our v4.2.3 except for some controllers where it couldn't delete .datz files and in some cases where it couldn't download - issues we are expecting 4.2.4 would resolve for us. 

[image]

markKHA, 2021-03-04:
We have noticed the same timeout issue. It seems like the system is able to download and delete the datZ files, but hits the timeout exceptions in TurnOffAsc3LoggingOverSnmp and TurnOnAsc3LoggingOverSnmp.

It seems like these two functions run regardless of controller type, even though the function calls out ASC3. My understanding is that without toggling the snmp flag off and on to force the controller to generate a new dat/datZ file, we're going to start missing some data. Is that specific to ASC3 controllers, or Cobalt and EOS as well?

NDSU-UGPTI-ATAC, 2021-03-23:
It appears to be non controller specific as we are unable to download data from any of them. We are able to ping them, login, download, and delete .datZ data via other programs (e.g. WinSCP)

ZachParkerCR, 2021-11-02:
> It appears to be non controller specific as we are unable to download data from any of them. We are able to ping them, login, download, and delete .datZ data via other programs (e.g. WinSCP)

@NDSU-UGPTI-ATAC , did you find a solution to this issue?  I'm running into a similar problem, while still on version ATSPM 4.2, with Cobalt controllers running on firmware 32.67.30.  The ftpfromallcontrollers is not able to connect to these controllers using normal FTP on port 21 and I cannot find a way to adjust ATSPM to connect on SFTP port 22.

ZachParkerCR, 2021-11-02:
@NDSU-UGPTI-ATAC , out of curiosity - what is the manufacturer and firmware version of the controllers that you cannot connect to with the ATSPM system?  I'm unable to connect to Econolite Cobalt controllers on firmware version 32.67.30 while running ATSPM 4.2.  

Would you be willing to share your batch file with me so I can review your commands and process?

ZachParkerCR, 2021-11-03:
@NDSU-UGPTI-ATAC , sorry to bug you, but I'm planning to update our environment to 4.24 and I don't want to run into the same issue after all the time spent performing the upgrade.  If you could let me know the firmware version and manufacturer of the controllers you are still having issues with, that will help me compare it to my environment and explain the situation to my traffic management so their expectations are aligned.

NDSU-UGPTI-ATAC, 2021-11-03:
@ZachParkerCR Econolite Cobalt Controllers running EOS.

NDSU-UGPTI-ATAC, 2021-11-04:
@ZachParkerCR I have sent you sample batch and text files. We will be testing 5.0 version's ftpfromallcontrollers program when released before upgrading to make sure the sftp works on our controllers.

ZachParkerCR, 2021-11-04:
> @ZachParkerCR I have sent you sample batch and text files. We will be testing 5.0 version's ftpfromallcontrollers program when released before upgrading to make sure the sftp works on our controllers.

@NDSU-UGPTI-ATAC , thanks again for all your offline help!