udotdevelopment/ATSPM #107: Is version 5.0 in development?

ZachParkerCR, 2022-01-06:
I heard that version 5.0 of the software is in development to be released soon.  I don't want to upgrade my version from 4.2 to 4.24 if 5.0 is on the horizon.  Is there an estimated release date the developers can let us know about?

Thank you!

taylormd3, 2022-01-07:
Zach, It's Mark Taylor at UDOT. This is a good question so let me be as specific as I can with version 5.0:

1) 5.0 is a big release (bigger then we generally like to do with releases). We don't have a release date yet but we're getting closer. I know it's a bit vague but we're getting very close to being done with two very large projects within version 5.0. a) Left Turn Gap Analysis Report and b) Data Compression. A third very large project (15-minute data aggregation files) has already been completed.

2) If I were to guess on an anticipated date (and it is a guess), I would think that 5.0 will be released probably around early spring. If you're already at version 4.2, I would just wait for 5.0 as there are so many exciting things coming with this new build as mentioned in A&B above. In addition, the developers are currently working on the following with version 5.0:

* Fix the "Timing and Actuation" metric so the pedestrian detectors are always correct that show on the display.
* Adding the new GDOT/Kimley Horn metic "Wait Time" pull request
* Adding the new MioVision (Traffops) pull request "Enhanced Configuration.
* Adding the new PennDOT "Methods and Assumptions" manual
* Adding a new link to PennDOT ATSPM Interface (https://www.dot.state.pa.us/public/Bureaus/BOMO/Portal/ATSPM/index.html)
* Fixing "Aggregate Data" report so the 15-minute aggregation tables are the data source.
* Moving the "Approach Speed" from the detector setting to the approach setting.

There are probably a few other things that will get done in version 5.0 (if not 5.0 probably 5.1 shortly thereafter):
* Add "Options and Thresholds" configuration page to the admin menu so owners can customize the default values without having to change the source code.
* Create 'Area and Agency" categories that will also be added to the maps.

If you have a functioning ATSPM system, I would suggest to wait for 5.0. It is actively being worked on by two developers. I promise that in the future, we will do much smaller releases so sorry about the long wait.

-- Mark Taylor

ChesireFreeThrow, 2022-02-16:
I'm moving this over to discussions since it's not an issue