udotdevelopment/ATSPM #184: 4.3 Release WCF Service missing dependencies

ian-cameron, 2022-09-19:
The MOEWcfServiceLibrary bin folder from 'ATSPM Executables' in the 4.3 Release is missing some packages.  Copying the executables results in service errors:

Could not load file or assembly 'Microsoft.Extensions.Caching.Abstractions' or one of its dependencies. The located assembly's manifest definition does not match the assembly reference. (Exception from HRESULT: 0x80131040)

Building the project in Release mode results in 104 files in the bin folder but the bin folder in the 'ATSPM Executables' only has 67 files.  The zip needs to include these dependencies or else people won't be able to use it to upgrade or install.

avetraffic, 2022-09-28:
I'm getting 67 files on when I build it in release mode. Can you give me the settings you use when building it? 
UDOT's current build should match the release, and it seemed to work on my personal computer as well.

ian-cameron, 2022-09-30:
Yep my mistake.  I think it was related to having parts from the old web.config file from the prior release, with old assembly references.  

As for my build folder having more files, I just realized I had previously built 4.2 a few months ago, and never cleaned before building... so there were leftovers mixed in with the new files, which happened to satisfy the old references from the old web.config to make my errors go away.

Sorry for the distraction.

closed 2022-09-30 by ian-cameron