udotdevelopment/ATSPM #222: ConvertDBForHistoricalConfigurations fails: could not load MOE.Common, an attempt was made to load a program with an incorrect format

dbrandesky, 2023-06-29:
I'm attempting to upgrade a 4.2.4 system to 4.3.1, but I can't get past the first step. ConvertDBForHistoricalConfigurations.exe is failing because of a System.BadImageFormatException. I experienced something similar to this with the website when I first launched it, then found I had to enable 32-bit support in the application pools in IIS. This fixed the website, but I don't know of an equivalent for this standalone exe.

System information:

- Server: Windows 2016 x64
- MS SQL: Express 2019
- Current ATSPM Version: 4.2.4

Full error message:

```
C:\Users\Administrator\Downloads\ATSPM.Executables_4.3.1\ATSPM Executables\ConvertDBForHistoricalConfigurations\bin\Release>ConvertDBForHistoricalConfigurations.exe

Unhandled Exception: System.BadImageFormatException: Could not load file or assembly 'MOE.Common, Version=2.16.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. An attempt was made to load a program with an incorrect format.
   at ConvertDBForHistoricalConfigurations.Program.RunMigrations()
   at ConvertDBForHistoricalConfigurations.Program.Main() in C:\Temp\projects\avenue\ATSPM\ConvertDBForHistoricalConfigurations\Program.cs:line 53
```

Looking at the source, line 53 is where `RunMigrations()` is called, which seems to be the primary function that calls other functions in the script.

dloweAvenueConsultants, 2023-06-30:
It looks like something got corrupted on this during the build for our deploy. I tested this one on my machine and it ran. Can you give it a try and see if it works?
[ConvertDBForHistoricalConfigurationsUpdated.zip](https://github.com/udotdevelopment/ATSPM/files/11920109/ConvertDBForHistoricalConfigurationsUpdated.zip)

dbrandesky, 2023-06-30:
> It looks like something got corrupted on this during the build for our deploy. I tested this one on my machine and it ran. Can you give it a try and see if it works? [...]

Thanks for the new link, but the zip you included here is empty. Not sure if it's being filtered or if the source file is missing something.

dloweAvenueConsultants, 2023-06-30:
Sorry about that. This one should have files this time.
[ConvertDBForHistoricalConfigurationsUpdated.zip](https://github.com/udotdevelopment/ATSPM/files/11923539/ConvertDBForHistoricalConfigurationsUpdated.zip)

dbrandesky, 2023-06-30:
> Sorry about that. This one should have files this time. [ConvertDBForHistoricalConfigurationsUpdated.zip](https://github.com/udotdevelopment/ATSPM/files/11923539/ConvertDBForHistoricalConfigurationsUpdated.zip)

Cool, that one appears to have worked; no errors thrown. Should we keep this open to fix the issue in the build for the next release?