udotdevelopment/ATSPM #226: 4.3.1 - Signal Map / GenerateAddData

Lee-ATSPM, 2023-10-12:
I upgraded a system to 4.3.1, and the signal map pins, regions and center are not changing from the release configuration. 

I've confirmed the GenerateAddData component is working and i can see the AddData and AddRouteData files populate within the webpage scripts folder. Ive opened the AddData script and it contains the correct signal ID's and locations. Other parts of the webpage function correctly as I can add signals and see the correct signals under "signal list".

dloweAvenueConsultants, 2023-10-12:
Sometimes the maps get cached. You can try clearing the cache(ctrl+f5) while you are on the page with the map.

Lee-ATSPM, 2023-10-12:
yes, i tried that initially as well and still have the same result.

closed 2023-10-12 by Lee-ATSPM