udotdevelopment/ATSPM #229: Adding Pages to 4.3.1

Lee-ATSPM, 2024-02-15:
I want to add a page to my system. Adding the entry using the Menu Configuration works perfectly but when i add my chtml document to what i think is the right location (views>home folder) is gives a 404 error. Is there another place i need to reference in the new document or another place it should be stored?

dloweAvenueConsultants, 2024-02-15:
It depends on the controller and action you chose. Everything is name based. So, if your document is called doc1 and you put it in the home directory you need to add a function called doc1 to the home controller. Then in the menu you would add the controller of home and the action of doc1.

Lee-ATSPM, 2024-02-15:
I see. I think that's the part I am missing. How do i add a function to the home controller?